honkmaster/homebridge-mi-flower-care #2: Crashing on startup

djensenius, 2018-03-24:
I’m getting this error:

```[2018-3-24 14:17:46] ====================
[2018-3-24 14:17:46] ERROR LOADING PLUGIN homebridge-mi-flower-care:
[2018-3-24 14:17:46] Error: Cannot find module '../build/Release/binding.node'
    at Function.Module._resolveFilename (module.js:547:15)
    at Function.Module._load (module.js:474:25)
    at Module.require (module.js:596:17)
    at require (internal/module.js:11:18)
    at Object.<anonymous> (/usr/lib/node_modules/homebridge-mi-flower-care/node_modules/bluetooth-hci-socket/lib/native.js:3:15)
    at Module._compile (module.js:652:30)
    at Object.Module._extensions..js (module.js:663:10)
    at Module.load (module.js:565:32)
    at tryModuleLoad (module.js:505:12)
    at Function.Module._load (module.js:497:3)
[2018-3-24 14:17:46] ====================```

honkmaster, 2018-03-24:
Which Pi are you using? have you installed the required packages?

djensenius, 2018-03-24:
Pi 3. 

I’ve installed the required packages. 
Running as root. 

Wondering if it’s calling binding.node correctly or if I need to install that by source?

honkmaster, 2018-03-24:
Could you check

```
ln -s nodejs node
```

djensenius, 2018-03-24:
```root@raspberrypi:/usr/bin# ln -s nodejs node
ln: failed to create symbolic link 'node': File exists```

```root@raspberrypi:/usr/bin# node --version
v8.10.0```

honkmaster, 2018-03-24:
I am very new to node, and thus have no further idea.

honkmaster, 2018-03-24:
Ou could check issues from other projects such as:

https://github.com/reelyactive/pi-suite/issues/1
https://github.com/noble/bleno/issues/140

Maybe they help.

djensenius, 2018-03-24:
Found the problem was that I needed to do this:

`npm install --unsafe-perm -g bluetooth-hci-socket`

honkmaster, 2018-03-24:
Perfect. Will add this later to a FAQ.

orilpik, 2018-03-24:
@djensenius Thanks man! It works for me now!!

Bankaifan, 2018-04-21:
Ich bekomme einen ähnlichen Fehler beim Booten

[2018-4-21 23:42:05] ====================
[2018-4-21 23:42:05] ERROR LOADING PLUGIN homebridge-mi-flower-care:
[2018-4-21 23:42:05] Error: Cannot find module '../build/Release/binding.node'
    at Function.Module._resolveFilename (internal/modules/cjs/loader.js:548:15)
    at Function.Module._load (internal/modules/cjs/loader.js:475:25)
    at Module.require (internal/modules/cjs/loader.js:598:17)
    at require (internal/modules/cjs/helpers.js:11:18)
    at Object.<anonymous> (/usr/local/lib/node_modules/homebridge-mi-flower-care/node_modules/bluetooth-hci-socket/lib/native.js:3:15)
    at Module._compile (internal/modules/cjs/loader.js:654:30)
    at Object.Module._extensions..js (internal/modules/cjs/loader.js:665:10)
    at Module.load (internal/modules/cjs/loader.js:566:32)
    at tryModuleLoad (internal/modules/cjs/loader.js:506:12)
    at Function.Module._load (internal/modules/cjs/loader.js:498:3)
    at Module.require (internal/modules/cjs/loader.js:598:17)
    at require (internal/modules/cjs/helpers.js:11:18)
    at Object.<anonymous> (/usr/local/lib/node_modules/homebridge-mi-flower-care/node_modules/bluetooth-hci-socket/index.js:8:20)
    at Module._compile (internal/modules/cjs/loader.js:654:30)
    at Object.Module._extensions..js (internal/modules/cjs/loader.js:665:10)
    at Module.load (internal/modules/cjs/loader.js:566:32)
[2018-4-21 23:42:05] ====================

honkmaster, 2018-04-22:
Have you tried:

`npm install --unsafe-perm -g bluetooth-hci-socket`

Are you running HB as root?

Bankaifan, 2018-04-22:
Yes and yes. 

I have Now Set Up a new SD-Card and try installing using the App

honkmaster, 2018-10-10:
Is this issue still relevant?

hagenfenris, 2018-10-27:
Hi,
hab hier das gleiche Problem:
`pi@raspberrypi:~ $ homebridge
[2018-10-27 09:45:26] Loaded config.json with 4 accessories and 1 platforms.
[2018-10-27 09:45:26] ---
[2018-10-27 09:45:27] Loaded plugin: homebridge-config-ui-x
[2018-10-27 09:45:27] Registering platform 'homebridge-config-ui-x.config'
[2018-10-27 09:45:27] ---
[2018-10-27 09:45:27] ====================
[2018-10-27 09:45:27] ERROR LOADING PLUGIN homebridge-mi-flower-care:
[2018-10-27 09:45:27] Error: Cannot find module '../build/Release/binding.node'
    at Function.Module._resolveFilename (module.js:557:15)
    at Function.Module._load (module.js:484:25)
    at Module.require (module.js:606:17)
    at require (internal/module.js:11:18)
    at Object.<anonymous> (/usr/local/lib/node_modules/homebridge-mi-flower-care/node_modules/bluetooth-hci-socket/lib/native.js:3:15)
    at Module._compile (module.js:662:30)
    at Object.Module._extensions..js (module.js:673:10)
    at Module.load (module.js:575:32)
    at tryModuleLoad (module.js:515:12)
    at Function.Module._load (module.js:507:3)
[2018-10-27 09:45:27] ====================
[2018-10-27 09:45:27] Loading 1 platforms...
[2018-10-27 09:45:27] [Homebridge] Initializing config platform...
[2018-10-27 09:45:27] [Homebridge] Spawning homebridge-config-ui-x with PID 1632
[2018-10-27 09:45:27] Loading 4 accessories...

/usr/local/lib/node_modules/homebridge/lib/api.js:64
      throw new Error("The requested accessory '" + name + "' was not registered by any plugin.");
            ^
Error: The requested accessory 'mi-flower-care' was not registered by any plugin.
    at API.accessory (/usr/local/lib/node_modules/homebridge/lib/api.js:64:13)
    at Server._loadAccessories (/usr/local/lib/node_modules/homebridge/lib/server.js:286:42)
    at Server.run (/usr/local/lib/node_modules/homebridge/lib/server.js:87:38)
    at module.exports (/usr/local/lib/node_modules/homebridge/lib/cli.js:45:10)
    at Object.<anonymous> (/usr/local/lib/node_modules/homebridge/bin/homebridge:17:22)
    at Module._compile (module.js:662:30)
    at Object.Module._extensions..js (module.js:673:10)
    at Module.load (module.js:575:32)
    at tryModuleLoad (module.js:515:12)
    at Function.Module._load (module.js:507:3)
`

npm install --unsafe-perm -g bluetooth-hci-socket   hab ich schon laufen lassen.
Per hcitool lescan bekomm ich meine Sensoren auch angezeigt.
Da ich neu bin, was das ganze Homebridge-Thema angelangt kann ich leider nicht viel mehr Infos geben.

Bankaifan, 2019-01-28:
Yes.
It seems the problem is, that the node-bluetooth-hci-socket does not work with the current node releases.
It seems that the developer does no longer work on it.
https://github.com/noble/node-bluetooth-hci-socket/issues/84

huczas, 2019-02-21:
same problem for me, is there a workaround for this problem? 
node -v
v11.10.0
and when I try to install node-bluetooth-hci-socket 

```
sudo npm install --unsafe-perm -g bluetooth-hci-socket

> usb@1.5.0 install /usr/lib/node_modules/bluetooth-hci-socket/node_modules/usb
> node-pre-gyp install --fallback-to-build

node-pre-gyp WARN Using request for node-pre-gyp https download
node-pre-gyp WARN Tried to download(404): https://github.com/tessel/node-usb/releases/download/1.5.0/usb_bindings-v1.5.0-node-v67-linux-arm.tar.gz
node-pre-gyp WARN Pre-built binaries not found for usb@1.5.0 and node@11.10.0 (node-v67 ABI, glibc) (falling back to source compile with node-gyp)
```
no clue how to install this plugin :(

Nastras, 2019-02-24:
For me the same Problem with Node 10. 
sudo npm install --unsafe-perm -g bluetooth-hci-socket it does not work -.-

honkmaster, 2019-03-18:
Node 10 is not supported atm. Please submit a PR if you need Node 10 or higher. You can orient yourself on https://github.com/hannseman/homebridge-mi-hygrothermograph. I do not have time to provide / implement the required changes.

honkmaster, 2020-07-12:
Closed as the plugin now supports the latest node.

closed 2020-07-12 by honkmaster