honkmaster/homebridge-mi-flower-care #4: No readings while using 'homebridge-mi-hygrothermograph' at the same time

wilkinsm89, 2018-04-08:
Works great when I run this plugin on it's own, however I get no readings in 'mi-flower-care' while also using 'homebridge-mi-hygrothermograph' at the same time.

Is anyone else running both able to get a reading?

honkmaster, 2018-04-09:
Both plugins use the library (noble)[https://github.com/noble/noble] in the background. I have no idea, why this libraries can not accessed by two plugins at the same time.

maltetgh, 2018-04-14:
I have the same problem.
Tried running both in separate instances but still no luck, readings in homebridge-mi-hygrothermograph worked but homebridge-mi-flower-care started to show crazy readings for my plants.

Will stop using both on the same hardware as seemingly multiple Plugins accessing BTLE is trouble.

I would love for this plugin to support the Xiaomi Mi Bluetooth Temperature and Humidity Sensor, because you already have a graphing/persistance for the flowers :)
@honkmaster Sag bescheid, wenn ich Dir meine Hardware zum testen leihen soll (den Mi Bluetooth Temperature and Humidity Sensor mit Display).

honkmaster, 2018-04-14:
I have no idea what is going on the the underlying plugin noble. It seems like the issue is [known](https://github.com/noble/noble/issues/260), but the "solution" is not satisfactory.

> @mcapraro you can only do this via separate Node.js processes for now, see https://github.com/sandeepmistry/noble#multiple-adapters

@maltetgh Thanks for the offer. I am currently very busy at work and have no time to go deep into node / JS and debugging.

yshinig, 2019-02-25:
Решили эту проблемму???

NathanAMiller, 2019-08-15:
Same issue here. I have 2 hygrothermograph and those work fine, but no data from flower care

NathanAMiller, 2019-08-15:
Also, when i set the Interval in this plugin, it override on Hygrothermograph plugin and it refresh as it set in Flower Care config!!!