honkmaster/homebridge-mi-flower-care #6: Automation based on lux level works only after opening Home app

piotrbielaska, 2018-05-22:
Hi. First I would like to thank you for great job on creating this plugin. Finally I and, what is most important, my wife has access to Mi Flower readings without need to open Mi Home app.

I did an automation based on lux level readings from the Mi Flower sensor which turns on and off the lights in the living room. It works but the problem is that automation happens only after I open Home app and readings are refreshed. Is it a common thing for current version of Homekit/Homebridge or a bug? 

In a mean time, I'm waiting for the shipment of  Mi Aqara Presence detector, which also have lux level sensor, and I hope I will be able to use it for the same purpose.

piotrbielaska, 2018-05-22:
You can use either [Hesperus](https://itunes.apple.com/us/app/hesperus/id969348892?mt=8), [Controller](https://itunes.apple.com/us/app/controller-for-homekit/id1198176727?mt=8&at=1000lHC8) or [Home by Matthias](https://itunes.apple.com/us/app/home-smart-home-automation/id995994352) iOS apps. They allow to use lux level as a trigger to start automation. It is great that you can set not only a value but also a range between values in which automation should be triggered.

paqpaqpaq, 2018-06-17:
same here
lux and moisture thresholds only work when an app is addressed that contains homekit setup.

deadlyknapsack, 2018-09-15:
> Hi. First I would like to thank you for great job on creating this plugin. Finally I and, what is most important, my wife has access to Mi Flower readings without need to open Mi Home app.
>
> I did an [...]

I am experiencing the same thing. Did you happen to find a solution? Built an automation, which works fine. However, only when I open the app myself the humidity values get updated in Home app.

piotrbielaska, 2018-09-15:
I think that there is no solution for this issue. I guess it's a matter of what automations are possible with HomeKit. I decided to start from scratch and use Home Assistant with HomeKit plugin instead. I have a database with history as additional feature. For automation, I use Node-Red. This way I was able to use Mi Flora as a source of lighting level to turn on/off my evening scene. It works perfectly.

honkmaster, 2018-10-09:
Coild you check the recent development (beta). The values are updated in the background:

npm i -g --unsafe-perm honkmaster/homebridge-mi-flower-care#async

honkmaster, 2018-10-10:
So, in my own experiment, automation in the background is working. I set a light (on / off), when lux falls below / rises above a certain level. Now it is your turn.

PR #11 (2018-10-11): Update of values in the background. 
Should fix #6 and #10.
commit e40ba9b6 (2018-10-11): Update of values in the background.  (#11)

* Fixes #6 and #10.
* Update dependencies (e.g. fakegato 0.5.2)
closed 2018-10-11 by honkmaster