honkmaster/homebridge-mi-flower-care #10: Plugin not updating values in background

deadlyknapsack, 2018-09-15:
I noticed that the plugin, regardless of the settings in config.json, is not getting value updates to Homebridge in the background. Is that deliberate? I am looking at the Home app on my iPad and the humidity percentage remains unchanged. The very second I open the Home app on my iPhone, the values are synced. Normally, that wouldn't be that big of a deal. However, I wanted to create an automation where a notification is sent when the humidity drops below a certain percentage. That becomes useless in terms of 'automation' when I have to get the percentage myself manually. 

Is this a bug or can you perhaps consider an enhancement on the plugin? I have about 15 Mi Flower modules and the idea was to have triggers when I need to water them... and now I am sad...

honkmaster, 2018-10-09:
Could you try a recent development?

npm i -g --unsafe-perm honkmaster/homebridge-mi-flower-care#async

honkmaster, 2018-10-10:
So, in my own experiment, automation in the background is working. I set a light (on / off), when lux falls below / rises above a certain level. Now it is your turn.

deadlyknapsack, 2018-10-11:
> So, in my own experiment, automation in the background is working. I set a light (on / off), when lux falls below / rises above a certain level. Now it is your turn.

Brilliant. Works like a charm. Haven't tested it thoroughly yet, but I can see the values being updated without me having to active in the app on my iPad or iPhone. To me, it is fixed. Thanks!

honkmaster, 2018-10-11:
Great feedback. Thanks. Will merge and publish to NPM this evening.

deadlyknapsack, 2018-10-11:
> Great feedback. Thanks. Will merge and publish to NPM this evening.

Should I change something in my setup? I have now installed the #async, but ideally would like to run the public version, no?

closed 2018-10-11 by honkmaster

honkmaster, 2018-10-11:
You can go back to the public version using:

`(sudo) npm install -g --unsafe-perm homebridge-mi-flower-care
`

Thanks again for testing.