honkmaster/homebridge-mi-flower-care #13: no data

BreitiDE, 2019-02-07:
hi,
I have installed homebridge on a raspberry, the Flower Care devices are shown in homebridge but data not shown. data keeps still 0 

#uname -a
Linux raspberrypi 4.14.79-v7+ #1159 SMP Sun Nov 4 17:50:20 GMT 2018 armv7l GNU/Linux

#node -v
v9.11.2

#npm -v
6.7.0

#npm -g list |grep homebridge
├─┬ homebridge@0.4.46
├─┬ homebridge-config-ui-x@3.9.6
├─┬ homebridge-hue@0.11.13
│ ├─┬ homebridge-hue-utils@2.0.9
│ │ ├── homebridge-lib@4.0.5 deduped
│ ├─┬ homebridge-lib@4.0.5
├─┬ homebridge-sonos@0.2.4

# npm -g list |grep bluetooth
├─┬ bluetooth-hci-socket@0.5.1

Viper-Gtr, 2019-03-07:
hi,
same problem here.
The configuration is exactly the same. No other plugin installed. 

Only difference: node-9.9.0

BreitiDE, 2019-03-08:
oh, I forget to mention that I also went down to 8.10.x with same issuing result ...

Hulliee, 2019-03-18:
Hi,
I'm experiencing the same with node 11.12.0. RPi 3

AndreiFAD, 2019-03-21:
hi

with RPI Zero W works fine.
but first I have connect with mobile Flower Care app and update device version v2.6.5 to v3.1.9

pi@raspberrypiW:~ $ uname -a
Linux raspberrypiW 4.14.79+ #1159 Sun Nov 4 17:28:08 GMT 2018 armv6l GNU/Linux
pi@raspberrypiW:~ $ node -v
v9.9.0
pi@raspberrypiW:~ $ npm -v
5.6.0
pi@raspberrypiW:~ $ npm -g list |grep homebridge
├─┬ homebridge@0.4.46
├─┬ homebridge-dht-sensor@2.1.2
├─┬ homebridge-gpio-device@0.4.3
├─┬ homebridge-mi-flower-care@1.1.0
├── homebridge-raspberrypi-temperature@0.0.7

omnidepp, 2019-05-25:
Same problem - devices are visible in Home and ElgatoEve, hcitools finds the sensors, but no data ever shows up.

With default logging  I get:

```
May 25 19:11:34 homeserver systemd-journald[261]: Suppressed 1656 messages from homebridge.service
May 25 19:11:34 homeserver homebridge[1023]: noble warning: unknown peripheral c47c8d6364b3

```
If I turn on debugging I get a lot of these:

```
May 25 19:06:19 homeserver homebridge[958]: Sat, 25 May 2019 17:06:19 GMT miflora device discovered:  undefined
May 25 19:06:20 homeserver homebridge[958]: Sat, 25 May 2019 17:06:20 GMT hci onSocketData: 043e2802010000b364638d7cc41c020106030295fe141695fe7120980085b364638d7cc40d041002a000b7
May 25 19:06:20 homeserver homebridge[958]: Sat, 25 May 2019 17:06:20 GMT hci   event type = 4
May 25 19:06:20 homeserver homebridge[958]: Sat, 25 May 2019 17:06:20 GMT hci   sub event type = 62
May 25 19:06:20 homeserver homebridge[958]: Sat, 25 May 2019 17:06:20 GMT hci           LE meta event type = 2
May 25 19:06:20 homeserver homebridge[958]: Sat, 25 May 2019 17:06:20 GMT hci           LE meta event status = 1
May 25 19:06:20 homeserver homebridge[958]: Sat, 25 May 2019 17:06:20 GMT hci           LE meta event data = 0000b364638d7cc41c020106030295fe141695fe7120980085b364638d7cc40d04100
May 25 19:06:20 homeserver homebridge[958]: Sat, 25 May 2019 17:06:20 GMT hci                   type = 0
May 25 19:06:20 homeserver homebridge[958]: Sat, 25 May 2019 17:06:20 GMT hci                   address = c4:7c:8d:63:64:b3
May 25 19:06:20 homeserver homebridge[958]: Sat, 25 May 2019 17:06:20 GMT hci                   address type = public
May 25 19:06:20 homeserver homebridge[958]: Sat, 25 May 2019 17:06:20 GMT hci                   eir = 020106030295fe141695fe7120980085b364638d7cc40d041002a000
May 25 19:06:20 homeserver homebridge[958]: Sat, 25 May 2019 17:06:20 GMT hci                   rssi = -73
May 25 19:06:20 homeserver homebridge[958]: Sat, 25 May 2019 17:06:20 GMT gap advertisement = {"localName":"Flower care","serviceData":[{"uuid":"fe95","data":{"type":"Buffer","da
```

### Environment

```
uname -a
Linux homeserver 4.15.0-50-generic #54-Ubuntu SMP Mon May 6 18:46:08 UTC 2019 x86_64 x86_64 x86_64 GNU/Linux

 npm -g list | grep homebridge
├─┬ homebridge@0.4.49
└─┬ homebridge-mi-flower-care@1.1.0

npm -v
3.5.2

node -v
v8.10.0
```

tonyduanesmith, 2019-07-24:
Did anybody work this out? I'm here with the same problem.

NathanAMiller, 2019-08-15:
Do you have "mi hygrothermograph" plugin installed?