honkmaster/homebridge-mi-flower-care #14: Tolerance range for the minHumidityAlert

flobernd, 2019-03-02:
Hi there,

could you please implement a "tolerance range" for the `minHumidityAlert` value? The sensor data fluctuates a little which causes the alert to switch from "Off" to "On" and vice versa multiple times within a few minutes. This is somewhat annoying if you have Homekit notifications enabled.

A tolerance range from e.g. 1 with a `minHumidityAlert` of 30 could cause the alert to go "On", if the value is 29 and only go "Off" if the value is up to 31.

Best regards

honkmaster, 2019-03-18:
I do not have time to provide / implement the requested changes. However, I am happy to accept a pull request.

honkmaster, 2022-08-21:
No feedback and PR, close the issue for now.

closed 2022-08-21 by honkmaster