honkmaster/homebridge-mi-flower-care #16: error installing on node 10.8.0

skop1986, 2019-04-30:
When installing the plug-in error appears on node 10.8.0, how can I get around the error? Please any help!

In file included from ../src/BluetoothHciSocket.cpp:8:0:
../../nan/nan.h:980:46: note: declared here
   NAN_DEPRECATED inline v8::Local<v8::Value> MakeCallback(
                                              ^~~~~~~~~~~~
binding.target.mk:99: ошибка выполнения рецепта для цели «Release/obj.target/binding/src/BluetoothHciSocket.o»
make: *** [Release/obj.target/binding/src/BluetoothHciSocket.o] Ошибка 1
make: выход из каталога «/usr/lib/node_modules/homebridge-mi-flower-care/node_modules/bluetooth-hci-socket/build»
gyp ERR! build error 
gyp ERR! stack Error: `make` failed with exit code: 2
gyp ERR! stack     at ChildProcess.onExit (/usr/lib/node_modules/npm/node_modules/node-gyp/lib/build.js:262:23)
gyp ERR! stack     at ChildProcess.emit (events.js:182:13)
gyp ERR! stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:238:12)
gyp ERR! System Linux 4.14.52-v7+
gyp ERR! command "/usr/bin/node" "/usr/lib/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js" "rebuild"
gyp ERR! cwd /usr/lib/node_modules/homebridge-mi-flower-care/node_modules/bluetooth-hci-socket
gyp ERR! node -v v10.8.0
gyp ERR! node-gyp -v v3.8.0
gyp ERR! not ok 
npm WARN optional SKIPPING OPTIONAL DEPENDENCY: xpc-connection@0.1.4 (node_modules/homebridge-mi-flower-care/node_modules/xpc-connection):
npm WARN notsup SKIPPING OPTIONAL DEPENDENCY: Unsupported platform for xpc-connection@0.1.4: wanted {"os":"darwin","arch":"any"} (current: {"os":"linux","arch":"arm"})
npm WARN optional SKIPPING OPTIONAL DEPENDENCY: bluetooth-hci-socket@0.5.1 (node_modules/homebridge-mi-flower-care/node_modules/bluetooth-hci-socket):
npm WARN optional SKIPPING OPTIONAL DEPENDENCY: bluetooth-hci-socket@0.5.1 install: `node-gyp rebuild`
npm WARN optional SKIPPING OPTIONAL DEPENDENCY: Exit status 1

+ homebridge-mi-flower-care@1.1.0
updated 1 package in 115.346s

honkmaster, 2019-05-10:
I am still running node 9 due to known issues. Will see later if these have already been resolved.

skop1986, 2019-05-16:
@honkmaster 

problem with bluetooth is resolved, this plugin works https://github.com/hannseman/homebridge-mi-hygrothermograph

honkmaster, 2019-05-16:
@skop1986 might be. Feel free to submit a PR, my time is currently very limited.

skop1986, 2019-05-16:
@honkmaster 

by cohabitation I am not a programmer and will not be able to participate

michaelzoellner, 2019-09-05:
> I am still running node 9 due to known issues. Will see later if these have already been resolved.

Could you tell me which node version it is exactly that you're using? I failed when I tried to install the plugin with node 9.11.2 as well as 8.11.1. It always fails at step 

`CXX(target) Release/obj.target/usb_bindings/src/node_usb.o`

I'm running the latest Raspbian on a freshly installed Raspberry Pi Zero W. The dependencies are installed correctly and I successfully scanned to receive the MAC address of the Flower Care sensor.

skop1986, 2019-09-05:
@michaelzoellner works for me on 9.11.2

michaelzoellner, 2019-09-05:
Hmm, did you install it just by 

`sudo npm install -g --unsafe-perm homebridge-mi-flower-care`

skop1986, 2019-09-05:
@michaelzoellner 

sudo apt-get install bluetooth bluez libbluetooth-dev libudev-dev

sudo npm install -g --unsafe-perm homebridge-mi-flower-care

michaelzoellner, 2019-09-06:
I tried once again from scratch with version 9.11.2, but no luck. This is what I always end up with:

```
pi@raspberrypi:~ $ sudo npm install -g --unsafe-perm homebridge-mi-flower-care
npm WARN notice [SECURITY] debug has the following vulnerability: 1 low. Go here for more details: https://www.npmjs.com/advisories?search=debug&version=2.2.0 - Run `npm i npm@latest -g` to upgrade your npm version, and then `npm audit` to get more info.

> usb@1.6.0 install /usr/local/lib/node_modules/homebridge-mi-flower-care/node_modules/usb
> prebuild-install --verbose || node-gyp rebuild

prebuild-install info begin Prebuild-install version 5.3.1
prebuild-install info looking for cached prebuild @ /root/.npm/_prebuilds/887a77-usb-v1.6.0-node-v59-linux-arm.tar.gz
prebuild-install http request GET https://github.com/tessel/node-usb/releases/download/v1.6.0/usb-v1.6.0-node-v59-linux-arm.tar.gz
prebuild-install http 404 https://github.com/tessel/node-usb/releases/download/v1.6.0/usb-v1.6.0-node-v59-linux-arm.tar.gz
prebuild-install WARN install No prebuilt binaries found (target=9.11.2 runtime=node arch=arm libc= platform=linux)
make: Entering directory '/usr/local/lib/node_modules/homebridge-mi-flower-care/node_modules/usb/build'
  CC(target) Release/obj.target/libusb/libusb/libusb/core.o
  CC(target) Release/obj.target/libusb/libusb/libusb/descriptor.o
  CC(target) Release/obj.target/libusb/libusb/libusb/hotplug.o
  CC(target) Release/obj.target/libusb/libusb/libusb/io.o
  CC(target) Release/obj.target/libusb/libusb/libusb/strerror.o
  CC(target) Release/obj.target/libusb/libusb/libusb/sync.o
  CC(target) Release/obj.target/libusb/libusb/libusb/os/poll_posix.o
  CC(target) Release/obj.target/libusb/libusb/libusb/os/threads_posix.o
  CC(target) Release/obj.target/libusb/libusb/libusb/os/linux_usbfs.o
  CC(target) Release/obj.target/libusb/libusb/libusb/os/linux_udev.o
  AR(target) Release/obj.target/usb.a
  COPY Release/usb.a
  CXX(target) Release/obj.target/usb_bindings/src/node_usb.o
In file included from ../src/node_usb.h:12,
                 from ../src/node_usb.cc:1:
/root/.node-gyp/9.11.2/include/node/v8.h: In static member function ‘static void v8::V8::RemoveGCPrologueCallback(v8::GCCallback)’:
/root/.node-gyp/9.11.2/include/node/v8.h:10378:53: warning: cast between incompatible function types from ‘v8::GCCallback’ {aka ‘void (*)(v8::GCType, v8::GCCallbackFlags)’} to ‘v8::Isolate::GCCallback’ {aka ‘void (*)(v8::Isolate*, v8::GCType, v8::GCCallbackFlags)’} [-Wcast-function-type]
       reinterpret_cast<Isolate::GCCallback>(callback));
                                                     ^
/root/.node-gyp/9.11.2/include/node/v8.h: In static member function ‘static void v8::V8::RemoveGCEpilogueCallback(v8::GCCallback)’:
/root/.node-gyp/9.11.2/include/node/v8.h:10385:53: warning: cast between incompatible function types from ‘v8::GCCallback’ {aka ‘void (*)(v8::GCType, v8::GCCallbackFlags)’} to ‘v8::Isolate::GCCallback’ {aka ‘void (*)(v8::Isolate*, v8::GCType, v8::GCCallbackFlags)’} [-Wcast-function-type]
       reinterpret_cast<Isolate::GCCallback>(callback));
                                                     ^
```

Many more errors follow, until at the end:

```
  SOLINK_MODULE(target) Release/obj.target/binding.node
  COPY Release/binding.node
make: Leaving directory '/usr/local/lib/node_modules/homebridge-mi-flower-care/node_modules/bluetooth-hci-socket/build'
npm WARN optional SKIPPING OPTIONAL DEPENDENCY: xpc-connection@0.1.4 (node_modules/homebridge-mi-flower-care/node_modules/xpc-connection):
npm WARN notsup SKIPPING OPTIONAL DEPENDENCY: Unsupported platform for xpc-connection@0.1.4: wanted {"os":"darwin","arch":"any"} (current: {"os":"linux","arch":"arm"})

+ homebridge-mi-flower-care@1.1.0
added 134 packages in 293.131s
```

Any clue? I wish I understood more about all of this... :-/

honkmaster, 2020-06-07:
Just update the submodules to support Node >= 10.

closed 2020-06-07 by honkmaster