honkmaster/homebridge-mi-flower-care #17: Feature Request: Disable Attribute

JangoBritt, 2019-05-10:
Hi!

Would it be an easy change, to allow the plugin to choose from the config.json to disable any of the elements not wanted in HomeKit? like Temperature, for example?

"DisableAttributes" : ["Temperature","Illumination"],

I've got three plants in the garden, for example, that are all giving me the same temperature reading, and it's not really adding any value to my already crowded collection of tiles.

honkmaster, 2019-05-10:
Feel free to add a PR.

JangoBritt, 2019-05-10:
I'm afraid I'm being quite cheeky with this request, because I've never used GitHub for development before.

I have experience with JavaScript, but not in Node.js, and not with GitHub. I could try, but not sure where to start. lol

JangoBritt, 2019-05-10:
Would this do it? What would happen if it wasn't added to the JSON?

    if (config.DisabledAttributes.Temperature = null ) {
        this.tempService = new Service.TemperatureSensor(this.name);
        this.tempService.getCharacteristic(Characteristic.CurrentTemperature)
            .on('get', this.getCurrentTemperature.bind(this));
        this.tempService.getCharacteristic(Characteristic.StatusLowBattery)
            .on('get', this.getStatusLowBattery.bind(this));
        this.tempService.getCharacteristic(Characteristic.StatusActive)
            .on('get', this.getStatusActive.bind(this));
    }

honkmaster, 2022-08-21:
No feedback and PR, close the issue for now.

closed 2022-08-21 by honkmaster