honkmaster/homebridge-mi-flower-care #21: Stop reading data after few days

Martvvliet, 2019-08-02:
Hi,
The plugin works fine for 1-3 days but then it stops working. It just keeps the same values. After a homebridge restart it works again. All the other plugins work fine.
Do you happen to know what this could be.
Also, is it possible to add some minor filtering of the received values, as it sometimes reads a crap value. Just an median filter would be nice :) thanks for the cool plugin :)

souimanga, 2019-08-04:
Hi, same problems :)

mattiasottosson, 2019-08-20:
I have the same problem. Logs show the following:

`
noble warning: unknown handle 64 disconnected!
RangeError: Index out of range
at checkOffset (buffer.js:977:11)
at Uint8Array.Buffer.readUInt16LE (buffer.js:1023:5)
at Hci.onSocketData (/usr/local/lib/node_modules/homebridge-mi-flower-care/node_modules/noble/lib/hci-socket/hci.js:464:18)
at emitOne (events.js:116:13)
at BluetoothHciSocket.emit (events.js:211:7)
`

souimanga, 2020-05-23:
Up up up

honkmaster, 2020-06-07:
I have one instance running now more than 7 months without any issues or reading missing. Thus I con not reproduce the issue.

closed 2020-06-07 by honkmaster