honkmaster/homebridge-mi-flower-care #23: Mi-Flower not work with Mi-Hygrothermograph

NathanAMiller, 2019-09-08:
I have an issue working with "homebridge-mi-flower" and "homebridge-mi-hygrothermograph" together.

When i remove "mi-hygrothermograph" plugin, "mi-flower" works fine. But when both are active, "mi-flower" not getting any data from devices and it shows "0" for all flower sensors.

[image]

NathanAMiller, 2020-01-12:
Up!

souimanga, 2020-01-14:
I have the same issue, do you have a solution ?

esgie, 2020-04-02:
Same problem here.

Satsang081, 2020-05-07:
Genau das gleiche Problem habe ich auch, auch wenn ich zwei Homebridge Instanzen auf dem Raps laufen lasse. Gibt es einen Lösungsansatz bezüglich noble?