honkmaster/homebridge-mi-flower-care #29: Low battery very fast

souimanga, 2019-11-28:
Hi all,
I’ve noticed several times that the device doesn’t send data anymore. When I put a new battery, it’s working again. 
BUT the battery life is very short, maybe several weeks... is that normal ? 
Thks

yannoz, 2019-12-14:
same here, after installing the plugin, the battery drain in two weeks, interval is way too hight, should be in days by default not seconds

souimanga, 2019-12-14:
Ahhh thanks @yannoz 
I feel less alone ;)

honkmaster, 2019-12-14:
I feel sorry, to hear this issue. However, my batteries last about half a year with the default settings. Thus, I think i am not able to help.

souimanga, 2019-12-15:
Hi @honkmaster many thanks for your concern! 
It’s weird ! 
I will send all my version installed and settings, so you could maybe tell me what’s wrong ??

souimanga, 2019-12-15:
Flower Care - V 3.2.1
Node - V 9.11.2
Homebridge 
Homebridge mi-flower-carte 
Homebridge Hue

souimanga, 2019-12-21:
Is that possible to set measurements to only one scan per day ? 
I’ve again one battery empty...

Aekung, 2020-08-05:
My set up drains battery really fast as well, around 3 weeks in total. Really which there would be a way to fix this.

bondskin, 2021-01-09:
Same here. I guess the max limit of 600 is pulling data too often. 
My battery survives just 3-4 weeks. That's quite short

benschwarz, 2022-12-26:
@bondskin, @Aekung, @souimanga did you manage to overcome the battery drain issue? Do you have recommended `interval` settings?

Looking at the documentation, the `interval` default is `600` (seconds) (10 minutes). A better default might be `43200` (12 hours). That way, there won't be an regularly active bluetooth connection, which should save precious battery power.

honkmaster, 2022-12-26:
The interval is a for polling. It does not guarantee, that any data is received. If it is very long and the connection unstable, data will never show up.

bondskin, 2022-12-26:
The outside temperature seems to be a huge influencer for the life time of the battery.
I have the interval running at 3600 and it delivered results.