honkmaster/homebridge-mi-flower-care #30: Update to work with  Node 12.13.1 LTS

BobbySlope, 2019-12-17:
Please update the plugin to work with  Node 12.13.1 LTS

honkmaster, 2019-12-17:
Feel free to submit a pull request.

2lazy2chill, 2019-12-18:
@honkmaster - Did now a pull request, but i'm unsure, if its placed correctly...? https://github.com/lucavb/homebridge-mi-flora/pulls

honkmaster, 2019-12-18:
Wrong place. Here is the correct one:
https://github.com/honkmaster/homebridge-mi-flower-care/pulls

NathanAMiller, 2020-04-04:
its not work on 10.x either

2lazy2chill, 2020-04-05:
Check out [https://github.com/Martvvliet/homebridge-mi-flora-filtered]. After last Update it works for me with HOOBS (which has Node 12.13.1) :-).

Sungamm, 2020-04-08:
> 
> 
> Check out https://github.com/Martvvliet/homebridge-mi-flora-filtered. After last Update it works for me with HOOBS (which has Node 12.13.1) :-).

Hi, are you sure. In the NPM description they stated clear node 9.xx is necessary.

2lazy2chill, 2020-04-09:
> Hi, are you sure. In the NPM description they stated clear node 9.xx is necessary.

Yes, it's running on my HOOBS.

Sungamm, 2020-04-09:
> 
> 
> > Hi, are you sure. In the NPM description they stated clear node 9.xx is necessary.
> 
> Yes, it's running on my HOOBS.

Thank you! Your are right and it is a great tipp! I installed the recommended version and it works with Node LTS. Thank you! :-)

honkmaster, 2020-06-07:
Just update the submodules to support Node >= 10.

closed 2020-06-07 by honkmaster