honkmaster/homebridge-mi-flower-care #31: HOOBS certification

BobbySlope, 2020-06-07:
hi @honkmaster we like to certify your plugin for HOOBS
please contact us via plugindev@hoobs.org and send us your email dress to invite you to our Slack

Regards Bobby

honkmaster, 2020-06-07:
Please send me more details why I would want to certify the component and what the implications are (legally, ...). You may find a contact form or my Mail address on my personal website.

MiranoVerhoef, 2020-07-07:
Please add support for hoobs, Alot of people use this since its provides a much nicer out of the box experience. It makes configuration and finding of your pluging alot easier. Please contact hoobs about this. I just ran into issues when trying to install your plugin, since the depedencies are not installed, which is hard to install on hoobs

honkmaster, 2020-07-07:
This may sound harsh: HOOBS want this plugin on their platform. Thus, they are in charge.
I asked for a mail with details, because they are missing on their website. 
This mail I have never received. Therefore, I am currently not willing to invest any more time. Sorry.

BobbySlope, 2020-07-08:
@honkmaster please send me an email again to bobby@hoobs.org
i missed the comment above one you mentioned.
we really like to get this plugin working on HOOBS

honkmaster, 2022-08-21:
Close as I never received a mail from HOOBS. The developer should not have to take action if someone else wants to get a benefit for their own platform.

closed 2022-08-21 by honkmaster