honkmaster/homebridge-mi-flower-care #33: Will it work with more than 1 Flora Care Device?

Aekung, 2020-06-14:
Hi @honkmaster ,

The plugin is working good for me so far and thank you very much for your recent update. However, I'm planning on adding one more VegTrug device but I'm not sure if the plugin would work with multiple devices and if yes, what should I put in the config file in order to make all works together.

By the way, I noticed that the Light Level reported in Home App is way higher than what it is shown in Mi Home. For example, the mi home would show 6 - 11 lux while homekit report 200+

Thank you very much and looking forward to hearing from you.

honkmaster, 2020-06-18:
The plugins works with more than one device.
I will check th lux value, however, I don't see an obvious error right now.

Aekung, 2020-06-22:
Hello,

Sorry for the late reply. I missed the reply and have just noticed that you already gave the answer once when I came back to check the thread.

Anyway, what should I put to the config.json file in order to use multiple devices since the accessory block can't state multiple times in the config.json file with the same accessory name? Can you please give me and idea on how to do this so that I can try?

By the way, thanks for looking into the issue of the lux value. Really looking forward to the update.

Thank you.

honkmaster, 2020-07-12:
Just put multiple instances in your config and use different names.

closed 2020-07-12 by honkmaster