honkmaster/homebridge-mi-flower-care #38: Homebridge 1.3.0 / characteristic was supplied illegal value

bondskin, 2021-02-20:
With 1.3.0 of Homebridge, I see the following error message

This plugin generated a warning from the characteristic 'Current Ambient Light Level': characteristic was supplied illegal value: number 0 exceeded minimum of 0.0001. See https://git.io/JtMGR for more info.

nielsth, 2021-02-21:
The issue only occurs when Homebridge has started but there is no available from the sensor yet. Usually, the first scan is done the specified interval (in seconds) after the plug-in has loaded. Sometimes more, in case the sensor is on the edge of the bluetooth range and isn't picked up on every scan. 

A possible solution might be to load the sensor in HomeKit with a No Response state and not write zero values to it, until the first successful scan has taken place. Would that work?

hs75, 2021-04-22:
same issue here, despite sensors within 2-3m range

Hansjshx, 2021-05-03:
I get this error _"[homebridge-mi-flora-filtered] This plugin generated a warning from the characteristic 'Soil Fertility': characteristic was supplied illegal value: number 2069 exceeded maximum of 255."_ from Fertility. FW 3.3.1, latest Homebridge, latest Plugin.

honkmaster, 2022-08-21:
Working on a fix.

ApoFenomeno, 2024-05-12:
Hello @honkmaster, is there any update on the fix, yet?

On Homebridge v1.8.1 and homebridge-mi-flower-care@1.3.1-beta4 there is still no sensor data provided in the Apple Home app or the Eve app.

The following data is exposed:
- Humidity 0%
- Temperature 0%
- Light 0,0001 Lux
- Battery Level is always 0
- Firmware 0.0

The device is next to the homebridge.