honkmaster/homebridge-mi-flower-care #39: Exposed Battery Level of the sensor is always 0, Firmware is 0.0

Lepidopteron, 2021-04-16:
Homebridge v1.3.4
homebridge-mi-flower-care@1.2.1

When checking the sensor's data in the Eve App, the following data is exposed
Battery Level is always 0
Firmware 0.0

Thus the alert/notification on low battery is not working anymore.

Data seems to be retrieved:
```
[16/04/2021, 19:31:01] [TEST 1] Lux: 385, Temperature: 21.2, Moisture: 47, Fertility: 260
[16/04/2021, 19:31:01] [TEST 1] Firmware: 3.3.1, Battery level: 86
```

honkmaster, 2022-08-21:
No idea what I may do here.

ApoFenomeno, 2024-05-12:
Same problem here, the plugins seems to be broken?!

Homebridge v1.8.1
homebridge-mi-flower-care@1.3.1-beta4 (also tested with latest version)

When checking the sensor's data in the Apple Home app or the Eve app, the following data is exposed:
Humidity 0%
Temperature 0%
Light 0,0001 Lux
Battery Level is always 0
Firmware 0.0