honkmaster/homebridge-mi-flower-care #40: [homebridge-mi-flower-care] This plugin generated a warning from the characteristic 'Soil Fertility'

CampoES, 2021-05-10:
Hello,

i installed your plugin for the first time. The status of Homebridge shows this warning:

[homebridge-mi-flower-care] This plugin generated a warning from the characteristic 'Soil Fertility': characteristic was supplied illegal value: number 883 exceeded maximum of 255.

And the fertilityvalue is not visible in (my?) Homebridge (no button) and false in Eve-App (always 255, probably set by Homebridge, when the value is about or equal to 255).

Any solution possible?

Thank you for your good work on this plugin.

Regards Michael

andutgv, 2021-06-05:
Same here. Waiting for a solution.

honkmaster, 2021-06-05:
This is currently a rather unimportant error for me, as the characteristic is not primary. Feel free to submit a PR.

bennovitch93, 2024-03-22:
For me I don't see the Soil Fertility at all and the Soil Humidity is shown as Air Humidity - any fixes planned on this?
Would be amazing! Best!

bennovitch93, 2024-03-24:
@honkmaster - thanks for your quick response!
Definitely Team Eve! ;)
Problem is the values are also not generated on homebridge itself - see screenshot attached.
Can you help?

[image]

newspeer, 2024-04-04:
Hey, I‘m having the same issue. The soil fertility sensor won‘t show up in Homebridge. Also not in Eve or Home. Additionally, Eve won‘t let me select a plant. I can only select the various sensor types from the „Type“ menu in Eve an check their values individually. As meintioned, except soil fertility.

I‘ve reinstalled the plugin multiple times. No change.